Serhioromano/vscode-st #3: Lint

Serhioromano, 2018-05-04:
Нужно тут выписать все правила которые нужно проверять и их чекать. Можешь добавить в комменте что ты думаешь нужно проверять, я добавлю в список сюда.

- [ ] - Переменная ни где не используется
- [ ] - Переменная не определена
- [ ] - Объявление не закрыто `;`

konopkov, 2018-05-26:
- [ ] double-vars - variable declared twice
- [ ] unconnected-inout - VAR_IN_OUT should be connected when calling function block
- [ ] camelcase - enforce camelcase naming convention
- [ ] no-tabs - disallow all tabs
- [ ] quotes - enforce the consistent use of either double, or single quotes
- [ ] require-stdoc - function block in/out vars and usage should be documented
- [ ] space-before-blocks - enforce consistent spacing before code blocks

Serhioromano, 2022-01-24:
New same topic #41

kropochev, 2025-05-12:
List of all Error Messages and Warnings https://content.helpme-codesys.com/en/CODESYS%20Development%20System/_cds_struct_reference_errors.html