Serhioromano/vscode-st #4: Suggestions

Serhioromano, 2018-05-04:
Не совсем уверен что мы это осилим. Хотелось бы анализ и подчказывать какие есть переменные у фукционального блока. Что бы когда точку нажимаешь, выпадали именно эти переменные.

tkucic, 2019-02-25:
It would be usefull to add TODO highlighting. The it is easy to spot TODO comments. You can just paint all capital TODO words with a different color.

Python example:
[image]

Serhioromano, 2019-02-27:
That is possible I think. But as I know highlight should be for `TODO:`. Should it?

tkucic, 2019-03-25:
You could add the ":" in regex i guess?

Serhioromano, 2019-03-26:
This feature (TODO) has been already implemented.

romecas, 2019-07-24:
Auto indentation would be a very nice to have feature. I saw some related code but it's still in comments, let me know if I can support somehow :)

[image]

Serhioromano, 2019-07-25:
DO you mean auto indentation when Format code? Because when you enter code I think there is auto indentation.

romecas, 2019-07-28:
The functionality I was trying to describe is to automatically format/smart indent a whole file (or selection) upon request. For instance by using the shortcut ctr shift f

-- Sent from /e/ Mail

Serhioromano, 2019-07-30:
I get it. Support for built-in formatter is on its way. Right now there is a format option but it only converts a low case to uppercase for `true` and `false` keywords. But this is proof of concept functionality. The full formatter will be later.

What do you think of the latest outline support feature?

Serhioromano, 2020-09-13:
@romecas in new 1.8.0 I added formatter that uses VS Code API. SO you can use standard Ctrl+Shift+F keys. But for now it only capitalize some built in functions. But I have basic structure, and indentation is on the way.

SteveBooth-1, 2021-05-20:
Status on indentation?  Would be super nice to have  Ctrl+Shift+F work for indentation. I can't find anyone anywhere that can re-indent ST code.

Serhioromano, 2021-06-27:
> @SteveBooth-1  Status on indentation? Would be super nice to have Ctrl+Shift+F work for indentation. I can't find anyone anywhere that can re-indent ST code.

There are 2 ways to do it. 1 - is a proper way to create LSP (Language Server Protocol) and this will not only indent but navigate to declarations, implementation intellisence for FB input and output variables and so on...

But this is HUUUGE work. Фтв  right now I do not have time to even start it.

2 - to create small function that will do it alongside with spaces and capitalization. This might be implemented fast but at the end time become obsolete when LSP is done.

Very hapd to make desicion.