Serhioromano/vscode-st #12: suggestion: remove auto-inserts of other paired keywords, improve snippets

msftrncs, 2018-09-01:
Like #10, I would like to suggest getting rid of the auto inserts of most all keyword pairs.  Only happens when they are entered in all uppercase, but that's how I am used to typing them, even though CoDeSys will automatically upper case them for me.  This would include `PROGRAM`, `FUNCTION`, `STRUCT`, `TYPE` and `ACTION` at least that I have found.  I think they cause more problems.  Snippets are the way to go.  They could use some improvement though.  

Snippets prefixes should be unique for each snippet, so that a user can determine a precise input for a specific snippet.  For instance, 'func' is used for both `FUNCTION` and `FUNCTION_BLOCK`.  I know I can type 'func' and press tab and I will get the snippet for `FUNCTION`, but getting to `FUCNTION_BLOCK` will require navigation of the intellisense menu.  It should just be 'down' one, but that's not certain.  Instead a prefix of 'funcb' would be enough.  The same would hold for `FOR` and `FOR BY`, and all the `VAR`'s

Serhioromano, 2018-09-04:
I see. Yes snippets require improvements. May be even add some snippets. Could you try to submit a PR? I would be happy to accept your help.

Regarding automatic closing, I am not sure. I mean your are right the way you code is totally legit. But nevertheless it is personal preference. Some might like this feature. Unfortunately VS Code does not allow make a configuration option for that. I'll ask them and see what we can do.

Serhioromano, 2018-09-04:
Started a topic in Microsoft/vscode#57838. Let's see how it goes.

closed 2018-12-20 by Serhioromano