Serhioromano/vscode-st #20: Incorrect auto formatting.

david9991, 2020-09-09:
Function name or maybe other place contains a keywords will be auto formatted to Upper characters.
for example:
`FUNCTION test_return_exit` will be auto formatted to `FUNCTION test_RETURN_EXIT`.

Serhioromano, 2020-09-11:
I was working on new Formating algorithm utilizing VS Code built in API. I've release v1.8.0 of extension, that one should fix the issue and also formating is now standard VS Code formating menu option.

It does not do much, but at least there is new extension architecture. I'll advance it later when have more time.

Serhioromano, 2020-09-13:
@david9991 please report after update if issue goes away.

david9991, 2020-09-13:
@Serhioromano Yes, seems it never happen again.

david9991, 2020-09-15:
@Serhioromano Hello, there is still some issue here. Keywords in quoted text still be formatted.

[image]

Serhioromano, 2020-10-13:
Fixed in latest version 1.8.2

closed 2020-10-13 by Serhioromano