Serhioromano/vscode-st #21: Logical Operator Colouring

kyle-goodwin, 2020-10-28:
In some themes, defaults included, the logical operators are not coloured.

To solve this I do one of the following:

- Change `keyword.operator.logical.st` to `keyword.operator.logical.python` in `st.tmLanguage.json`.
- Add `keyword.operator.logical.st` to the same block with `keyword.operator.logical.python` in `dark_vs.json`.

The first solution is not correct because this is not a python extension. I'm not sure why python is the only one that gets called out for clouring in the theme.

The second one is more proper but only affects the 1 theme and isn't something the extension can solve on it's own.

Another ST extension uses `support.class.collection.st` which I don't think is correct either.

Pretty minor issue and I have no idea how to fix it properly but would be nice to get it highlighting in all themes.

Serhioromano, 2020-11-18:
One of the solutions is to create theme and install it with extension so that theme may support even more colorings.

May be it is possible to contribute to `dark_vs.json`? I created on
https://github.com/microsoft/vscode/pull/110847

Support it with a thumb up.

Serhioromano, 2020-11-19:
My PR was rejected but they gave some suggestions. Read it.

kyle-goodwin, 2020-11-19:
Bit of a lame reason for rejecting the request. It does seem like an oversight in that it is only implemented for Python. Oh well.

closed 2020-11-19 by kyle-goodwin