Serhioromano/vscode-st #23: An error message is prompted when executing the command of YARN

MRFengC, 2021-01-04:
$ node ./node_modules/vscode/bin/install
Detected VS Code engine version: 1.52.1
Found minimal version that qualifies engine range: 1.52.1
Fetching vscode.d.ts from: https://raw.githubusercontent.com/Microsoft/vscode/ea3859d4ba2f3e577a159bc91e3074c5d85c0523/src/vs/vscode.d.ts
Error installing vscode.d.ts: Error: read ECONNRESET
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/install for documentation about this command.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/install for documentation about this command.

Serhioromano, 2021-01-04:
What YARN? This is not a Yarn or NPM package. It is VS Code extension. Tell me mor about this error. Why you execute yarn? How that is connected to this extension and Structured Text? Where in the error you find that this error relates to my extension?

MRFengC, 2021-01-19:
> 什么纱？这不是Yarn或NPM软件包。这是VS Code扩展。告诉我关于这个错误的消息。为什么要执行毛线？它如何与此扩展名和结构化文本连接？您在错误中的哪里发现此错误与我的扩展名有关？

Sorry,The reason for my misconfiguration.

closed 2021-01-19 by MRFengC