Serhioromano/vscode-st #24: Request: Make TCP Autocompletion an Extension Option

mfgeorge, 2021-01-05:
Often times in robotics / automation the term TCP is used to abbreviate "Tool Center Point". 

However, this conflicts with the `TCP ... END_TCP` auto-completion. 

I notice that code is auto-created the instant I type `TCP` in all upper-case, regardless of what follows.
I am requesting that the auto-completion of `TCP ... END_TCP` be made optional. 

This would save me some sanity and would be greatly appreciated!

For example, I desire to type `TCP_X` (in general `TCP_<>`) without a `TCP ... END_TCP` block getting auto-created.

Thank you for your consideration. 
Thank you for your VS code extension. I really appreciate what you have created here!

Serhioromano, 2021-01-06:
This is not easy to do. I do not have control over it's behavior. It is configured here:

https://github.com/Serhioromano/vscode-st/blob/0a320b00d28c42935381b697f56a55c3f837b141/language-configuration.json#L78

I can only totally delete it.

mfgeorge, 2021-01-07:
OK - Thanks for the information. 

Sharing this information for any future user. To remove this functionality from an installed version of this extension: 
1. Navigate to your `$HOME/. vscode/extensions/serhioromano.vscode-st-<version>` folder
2. Open `language-configuration.json` in an editor and modify as detailed by @Serhioromano in last comment
3. Save file and re-open VS Code, Change to the language configuration should now be active

Please correct me if I am wrong. 

Due to the difficulty, no action needed on this issue. 

Thanks @Serhioromano !

closed 2021-01-07 by mfgeorge

Serhioromano, 2021-01-12:
Maybe I'll delete it for future versions. SO you will not need to update it after every update of extension.