Serhioromano/vscode-st #25: Symbol Provider unaware of comments/strings, missing many POU objects, limited breadcrumb support.

msftrncs, 2021-01-27:
The current `STDocumentSymbolProvider` (outline and breadcrumb feature) is unaware of comments and text strings, and will 'detect' symbols where none exists, and sometimes fails to detect them over small syntactical issues.  It doesn't include global VAR's in addition to some issues with VAR's in general (modifiers).  It also doesn't provide the data needed for breadcrumbs to work correctly.

It is a great start, though.  I have been working on improving it for a few days, and may have drastic improvements done in a few more days.

Serhioromano, 2021-01-27:
Thank you. This is more or less skillful work and require a lot of learning things. I had no time to look deep into VS Code Extension API. To make it work as you say, I think LSP should be implemented. 

> It is a great start, though. I have been working on improving it for a few days, and may have drastic improvements done in a few more days.

That would be perfect. I'll gladly accept your improvements. I'll go now to your fork and check your changes.

msftrncs, 2021-01-29:
While I have a couple of loose ends remaining, it could be taken a look at now.

https://github.com/msftrncs/vscode-st/tree/symbolprovider_ideas
https://github.com/msftrncs/vscode-st/blob/symbolprovider_ideas/src/symbolprovider.ts

The remaining things are dealing with modifiers on VAR blocks, and comma delimited variable lists in VAR blocks (and maybe STRUCT/UNION as well, which is okay, its the same `listVariables` function.

The thing to notice is that the breadcrumb navigator is fully able to indicate which POU element you are in from the cursor position.

Serhioromano, 2021-01-29:
Looking good. Can you upstream you `symbolprovider_ideas` branch from latest `master`? First it will make your work mergeable. Second in compare it will skip all changes from latest files.

msftrncs, 2021-01-31:
I've reached the point where I have formally submitted the revised SymbolProvider as a PR, which should ease evaluation.   I don't think it is ready to be done, but it certainly needs further review than what I can do with the kind of ST files I typically see.

Serhioromano, 2021-01-31:
Looking good. Could you recreate PR against develop branch please.

closed 2021-01-31 by Serhioromano