Serhioromano/vscode-st #29: Formatter issues

msftrncs, 2021-02-01:
The formatter (capitalizer) has a couple issues, and exposes a grammar issue.

```iecst
TYPE Bob : (* test *) STRUCT
        A, B : INT; //test
        b : INT; //test
    END_STRUCT
END_TYPE

(* Bool Word Struct Constant 
```
[image]

when formatted:
[image]

Note there are multiple issues. 

1. It capitalized a broken comment.  The comment doesn't end, and that lack of an end trips up the regex used to detect keywords, while ignoring comments and quoted strings.
2. It thought the start of the comment was a parameter list for a function call or operator `END_TYPE` so it moved it.  'keywords' and the 'end' block keywords should be excluded from this treatment.,
3. The TM grammar changed the scope of the keyword  `END_TYPE` when it has a parenthesis behind it.   Since  `END_TYPE` is a rigid keyword, it should appear very high in the processing in the grammar file, above anything that utilizes a more generic form.

 The `VAR_` keywords, but not `VAR` or `END_VAR`, seem to be missing from the capitalization behavior.  This is not shown above.

Serhioromano, 2021-02-01:
That is strange. Here is the code 

```js
let regEx = new RegExp(`\\b(?:${functions.join('|')})\\b\\s+\\(`, "ig");
```

It should move only functions. 

The problem was in `function` array. Last element was empty.

Serhioromano, 2021-02-01:
3. Looks like `VAR_*` is not considered at all. Also fixed that.

Serhioromano, 2021-02-01:
New version released 1.9.1 with 2 fixes.

commit 49be29ef (2021-02-01): Fixed #29
closed 2021-02-01 by Serhioromano