Serhioromano/vscode-st #30: 1.9.0/1.9.1 SymbolProvider regressions

msftrncs, 2021-02-02:
With the 1.9.0/1.9.1 SymbolProvider changes:
1. may take excessive time for malformed comments/quoted strings in a `VAR`(_x) block
2. fails to mark CONSTANT variables
3. may show nonexistent '<unnamed>' variables from final comments or from comments separated from previous variable by two line breaks.

I will post a PR shortly.

With #1 under control, I was able to open a 32,000 line CoDeSys EXP file, and have a presentation of the approximately 4,000 symbols in under 2 seconds.

msftrncs, 2021-02-02:
Side issue I haven't solved yet, a missing `END_VAR` between two `VAR`'s, causes the `VAR` keyword to be included as a variable in the previous `VAR`.

``` iecst
VAR
    hello : SumType;
(* END_VAR oops *)
VAR    (* <-- taken as a variable name with type SumType2 *)
    hello2 : SumType2;   (* hello2 variable will not appear in symbol list *)
END_VAR
```

Serhioromano, 2021-02-02:
Should this one be closed already?

msftrncs, 2021-02-02:
Yes, closed by #32 being merged.    (I didn't use the keyword 'fixes' when I mentioned issue)

closed 2021-02-02 by msftrncs