Serhioromano/vscode-st #31: VAR/END_VAR scope when as parts of other identifiers

msftrncs, 2021-02-02:
`VAR` and `END_VAR` incorrectly scope when run in to other identifiers in certain orders

[image]

`VAR` does not validate word break at end, `END_VAR` does not validate work break at either end.

`VAR_x` seem to be same as `VAR`

Looks like the \b is missing from the end of both the begin and end clauses.

Serhioromano, 2021-02-02:
If you put all in one line not sure it is fixable. Scope is defined with JSON file, and basically as I understand it, VS Code takes care of scopes, open\close brackets.

msftrncs, 2021-02-02:
https://github.com/Serhioromano/vscode-st/blob/f33afa298c99bbc60bc3b11f04945ef32df050f4/syntaxes/st.tmLanguage.json#L220-L226

```
"begin": "\\b(?:VAR(?:_(?:INPUT|OUTPUT|IN_OUT|TEMP|GLOBAL|ACCESS|EXTERNAL))?|STRUCT|UNION|PUBLIC|PRIVATE|PROTECTED)\\b", 
```
```
"end": "\\b(?:END_(?:VAR|STRUCT|UNION))\\b",
```

(different coloring here, but now the keywords run together are no longer 'keywords')
[image]

Serhioromano, 2021-02-02:
> (different coloring here, but now the keywords run together are no longer 'keywords')
> [image]

Is that how it supposed to behave? Looks correct to me. You can PR this change.

msftrncs, 2021-02-02:
After Appling correction:

[image]

PR #33 (2021-02-02): apply word breaks to VAR and END_VAR keyword detection.
Fixes #31, applying word break requirements to the keywords to insure they are not considered a keyword if run together with other text or keywords.
commit 6860c0f1 (2021-02-02): apply word breaks to VAR and END_VAR keyword detection.

Fixes #31
closed 2021-02-02 by Serhioromano