Serhioromano/vscode-st #34: quoted string character escapes in grammar

msftrncs, 2021-02-03:
The grammar has escape patterns for quoted strings, indicated as `\\\\.` in the grammar.  These would be C style escapes, and they do not seem to be part of the IEC 61131-3 standard.

For reference:
https://www.fernhillsoftware.com/help/iec-61131/common-elements/literals-string.html

Sequence | Usage | Description
-- | -- | --
$$ | Both | Dollar sign ($) character
$' | Single | Apostrophe (') character
$" | Double | Double quotation mark (") character
$L | Both | Line feed (U+000A) character
$N | Both | Line feed (U+000A) character
$R | Both | Carriage return (U+000D) character
$T | Both | Horizontal tab (U+0009) character
$nn | Single | Single byte character, where nn is the two digit hexadecimal value of the character
$nnnn | Double | Double byte character, where nnnn is the four digit hexadecimal value of the character

Upper or lowercase letter are allowed.

But I see no support for them in the grammar.  

Regarding implementation, from what I can tell, invalid combinations are not supported, and for instance in CoDeSys, they are flagged as an error.

Sample presentation (similar to CoDeSys)
[image]

Current grammar: (gets out of sync)
[image]

Do your environments actually support C style escapes, or is this a mistake?   This would be a concern for cross platform compatibility of code, if all systems do not utilize a single standard, there are already enough extensions to the standard to make things difficult.

These character escapes also appear in the comment sections of the grammar.  They could potentially cause a user unexpected issues:

```iecst
/*///////// this comment doesn't close \\\\\\\\\*/ oh no this is still comment...
```

[image]

I don't know of any systems that allow this, so I don't think this is correct.

Serhioromano, 2021-02-03:
> Do your environments actually support C style escapes, or is this a mistake?

Mistake.

Will you PR or should I fix myself?
What Thene do you use in VS Code?

Serhioromano, 2021-02-03:
I have fixed this in develop branch. I'll release a hotfix 1.9.4 now.

Serhioromano, 2021-02-03:
Done. New version is up. Changes in master.

Serhioromano, 2021-02-03:
And by the way, thank you very much for such an involvement!

closed 2021-02-03 by Serhioromano

Serhioromano, 2021-02-03:
I heavily refactored syntax highlights and fixed a lot of tiny mistakes. Including string escapes. v 1.10.0

msftrncs, 2021-02-05:
Thank-you.  I use a customized theme derived from Monokai Dimmed, its not released, but its available in https://github.com/msftrncs/theme-monokai-less-dimmed.   I have tweaked it pretty heavily to make all the languages I use all look generally the same even though they all use different scoping.   Its heavily colored.