Serhioromano/vscode-st #36: Suggestion, split view is VS code

r9guy, 2021-08-31:
Is it possible to split editor to 2 sections, like Visual Studio does?
This would be a step closer to separate Declaration from Definition.

slimeth, 2021-11-02:
@r9guy suggestion seems actually quite relevant and would be not only cosmetically beneficial. TwinCAT seems rather sensitive on that XML code on top, between and after the declaration and definition. I realized that after I activated the auto-formatting of VS code which also formatted the XML code and afterward TwinCAT couldn't read the file properly anymore.
[image]
[image]

r9guy, 2021-11-02:
Not only that but TwinCAT may change some UIDs in that XML portion and
version control system must not be fooled by that.

Serhioromano, 2021-11-10:
This is not easy to do for TwinCat. Because it stores files as XML files. Although perhaps that would make a lot of sense is files of this  extension is opened that way. Anyway right now I have no idea how I could pull this off.

closed 2021-11-10 by Serhioromano

r9guy, 2022-04-21:
This case is closed, however I have seen the following project from Microsoft.
[Interactive](https://github.com/dotnet/interactive). 
they also have in preview stage extension for VX2022. basically what they have done is create notebook like environment for any programming language.

Serhioromano, 2022-05-10:
I've seen this feature in VSC. But I still did not see a practical application for this case.