Serhioromano/vscode-st #37: Using formatter issue

ChrisNiklasLucka, 2021-11-09:
Hi serhioromano,

I am using your program and it works really nice, but I have an issue using the formatter. How can I set um, that <> isn't change to < >? 

Thanks at first and BR
Chris

Serhioromano, 2021-11-10:
Thank you. I have fixed in in next update by the end of this week.

closed 2021-11-10 by Serhioromano

ChrisNiklasLucka, 2021-11-17:
Hi serhioromano,

great work, can you publish this extension with all updates to vs code extensions? 

Would be great. Thanks!

Serhioromano, 2021-11-21:
I cannot. There is always an error. I still investigate what is going on.

ChrisNiklasLucka, 2021-11-22:
ok then I keep my fingers crossed that this will work out in the near future

Serhioromano, 2021-12-10:
@ChrisNiklasLucka 
Finally was able to publish 1.10.8 after full WSL reinstall.