Serhioromano/vscode-st #38: Issue with **

ChrisNiklasLucka, 2021-11-17:
Hi Serhioromano,

I've also found an issue with the ** which is automatically changed to * *. Is there also an option, that you fix this?

By the way I have more topics for your grandious extension but I don't know if I can ask here for all these stuff and give you tasks, but I can't realize an extension by my own. Is there any interest from your side to realize some more stuff in this extension or should it be an running version and thats it?

Thanks for all your help and best regards
Chris

Serhioromano, 2021-11-21:
Fixed/

Serhioromano, 2021-12-10:
Finally was able to publish 1.10.8 after full WSL reinstall.

ChrisNiklasLucka, 2021-12-16:
Great work @Serhioromano !
Did you read this: 
_By the way I have more topics for your grandious extension but I don't know if I can ask here for all these stuff and give you tasks, but I can't realize an extension by my own. Is there any interest from your side to realize some more stuff in this extension or should it be an running version and thats it? I read, that you want to realize an compiler, is there any update for this?

ChrisNiklasLucka, 2021-12-16:
Hi @Serhioromano tested your new update and the formatter did not work now, is there another bug? Thanks to you!

Serhioromano, 2022-01-07:
I'll test is now. You can create as many issues here as you wish, I'll be more than glad to solve all I can.

Serhioromano, 2022-01-07:
Released 1.11.1 version should work. I have fully refactored spaces formating.

closed 2022-01-07 by Serhioromano

ChrisNiklasLucka, 2022-01-07:
Hello @Serhioromano ** still changes to * *. Is there anything missing?

ChrisNiklasLucka, 2022-01-07:
@Serhioromano thanks for the great work (i deleted my last comment, it was wrong). Thank you very much!