Serhioromano/vscode-st #41: Need a Linter in this extension

ChrisNiklasLucka, 2022-01-17:
Hi @Serhioromano,
to get nice and structured text, I need a linter in this extension. So in best case I want some text in the VS console, when naming of a variable is not right. Don't know if you can implement it? If you can do it I can put all linting-Tasks in a tabular here. 

Best regards and stay safe
Chris

ChrisNiklasLucka, 2022-01-25:
Hi @Serhioromano thats great. Here you'll have a documents with requirements from my side - about 25 requirements (not all for linter I think).
BR from Hamburg
[Requirements_Linter.pdf](https://github.com/Serhioromano/vscode-st/files/7931562/Requirements_Linter.pdf)
g

Serhioromano, 2022-01-25:
@ChrisNiklasLucka this is very well written document. Thank you. Right now I am working on new Git Flow extension, but as soon as I finish it, I'll try to improve this one.

ChrisNiklasLucka, 2022-01-26:
@Serhioromano thanks! Do you have any "timeline" for your programming goals? Do you think that this linter is ready in ~1 month or is it more a task about a half year? I ask, just du get an idea how long it whould take.
I have not that much knowledge in this programming language, but in other languages like python I have enought knowlegde, so if you can create a task line or something like this i could support you with my manpower.

Serhioromano, 2022-01-26:
Right now, half year sounds more realistic since I am doing it alone. You see the point with lint is not only lint. In order to implement it LSP has to be created with full code parse capability. Lint will be just a byproduct of that. I mean to have lint a full blown [code parser](https://code.visualstudio.com/api/language-extensions/language-server-extension-guide) have to be created.

Serhioromano, 2022-06-29:
> @Serhioromano Probably you want to use [ATNLR4](https://www.antlr.org/download.html) for this task. I understand that this will be a complete re-write of the code, but a parser/lexer will work much better than a REGEXP based parser. You define the grammar for ST and generate the parser/lexer with ANTLR4. Unfortunately it will be in Javascript and not TypeScript, so I don't know if this will work as part of the VSCode extension which is based on TypeScript.

Thank you. I was looking into this lib for quite a while already. Unfortunately I cannot find a guide how to do that. Docs I've found was not good enough. It is written as for someone who already understand parser and lexer just need to know how ATNLR works. I could not get a clear picture. Although I've created ATNLR file for ST.

ThirtySomething, 2022-07-02:
> Thank you. I was looking into this lib for quite a while already. Unfortunately I cannot find a guide how to do that. Docs I've found was not good enough. It is written as for someone who already [...]

I agree with you about the documentation. As far as I understood the process consists of a few step. First of all you have to define your grammar. Then you have to create the lexer/parser based on this grammar. Then you have to hook into the generated code (or vice versa) to work with the lexer/parser result.

I'll try to figure out how to work with this, too. I want to use it for a simple project of mine and also later on for a much more complex project.

ThirtySomething, 2022-07-02:
You also need some kind of runtime. For JavaScript I suppose [this](https://github.com/antlr/antlr4/blob/master/doc/javascript-target.md) is a good point to start.

Serhioromano, 2022-07-04:
Ш was successful at creating grammar and then generate lexer and parser. The thing I do not completely understand how to turn it into LSP.

ThirtySomething, 2022-07-04:
> Ш was successful at creating grammar and then generate lexer and parser. The thing I do not completely understand how to turn it into LSP.

Oh, I understand. For this I'm not able to help. I'm struggling with my own project because it's in C++ and the ANTLR4 runtime is not compiling at the moment. But maybe [this repository](https://github.com/kaby76/AntlrVSIX) might throw some light on the topic. As far as I understood it's a plugin for VSCode using the LSP and supports ANTLR parse trees.

Serhioromano, 2022-07-05:
> Oh, I understand. For this I'm not able to help. I'm struggling with my own project because it's in C++ and the ANTLR4 runtime is not compiling at the moment. But maybe [this [...]

Thank you for the link. That di not make anything clear as it is in C#. Although I used to code in C# decade ago and can read it partially, there are still lot of questions.

I also read this article 
https://neuroning.com/post/implementing-code-completion-for-vscode-with-antlr/

After which I fill like stupid. How come he explained everything and I still cannot make it.

ThirtySomething, 2022-07-05:
Okay, as far as I undrestand either the LSP itself or the interaction between LSP and ANTLR is not clear. This is also true for me - how to integrate own logic to the ANTLR generated code. Some googling came up with:
- [LSP description](https://code.visualstudio.com/api/language-extensions/language-server-extension-guide)
- [Step By Step development](https://medium.com/codex/antlr-magic-developing-mainframe-language-applications-using-language-recognizer-5262726e1e93) 
- [Editor with ANTLR and LSP](https://betterprogramming.pub/create-a-custom-web-editor-using-typescript-react-antlr-and-monaco-editor-part-1-2f710c69c18c)
Unfortunately I got only basic skills in JS development and hope this will help.

JohnnyOpcode, 2022-09-09:
Have a look at this repo. There was an IEC compiler written quite some time ago.

https://github.com/thiagoralves/OpenPLC_Editor/tree/master/matiec

AndersHogqvist, 2022-09-25:
> I'm also looking toward golang gocc. This library is pure magic. It is the same concept you create BNF files and based on it parse it into parser, lexer and AST but as an outcome you can translate your piece of code to another language. Gor example convert ST file to C++ or use it as compiler to create executable directly.

Sorry for highjacking this issue but that last sentence sounds very interesting for another reason. I'm like into how to do an input filter for Doxygen in order to generate documentation from comments etc. I've read an article describing this process for turning structured text into C# pseudo code that Doxygen can interpret. If what you're saying is true, the part about converting ST to C++, that would be absolutely awesome since that is something Doxygen can read!

AndersHogqvist, 2022-09-25:
This might come in handy. It contains the EBNF för ST that I suppose can be used as input to gocc:
https://www.researchgate.net/publication/228971719_A_syntactic_specification_for_the_programming_languages_of_the_IEC_61131-3_standard

Serhioromano, 2022-09-26:
> This might come in handy. It contains the EBNF för ST that I suppose can be used as input to gocc: https://www.researchgate.net/publication/228971719_A_syntactic_specification_for_the_programming_languages_of_the_IEC_61131-3_standard

Thank you. That article indeed very useful. It will not work for GOCC, there is slightly different variation of BNF but it might serve as a base for sure.

AndersHogqvist, 2022-10-20:
I noticed that some language servers use tree-sitter so I looked into that. I have managed to create a grammar that successfully parses our source code and I'm currently looking into the syntax highlighting. It looks promising but it's still in a very very early experimental stage 🙂

Serhioromano, 2022-11-22:
> I also managed to implement the rules for parsing structured text using [chevrotain.](https://github.com/chevrotain/chevrotain) (example of what it is [here](https://chevrotain.io/playground/)) I see that @AndersHogqvist is doing something similiar. I'm not sure which would be better, but I would like to help with your linter effort.

This project looks awesome. Not only it is pure JS parser not a parser generator it can be used to create compiler too as long as LSP. Looks promising. I played with it but I stack with some issues which I cannot resolve. I posted an issue.

Serhioromano, 2022-11-23:
@rempAut  Look at this https://github.com/langium/langium

rempAut, 2022-11-24:
> @rempAut @AndersHogqvist Look at this https://github.com/langium/langium
> 
> Edited: My goodness!!! It is that simple!

It really looks like magic! I tried [expressions](https://gist.github.com/rempAut/294af3f12914ab55920aa0af88b619e9) in the playground too, it's mostly only re-writing the EBNF rules, and it works. This is very promising.

Serhioromano, 2022-11-24:
If you fork this repo and run `npm install` then you can see `LSP` folder. There is langium, xtext and chevrotain tests. If you run `node LSP/chevrotain.js` you will see result in a  terminal and html file with diagram will be generated.

Serhioromano, 2022-11-24:
I was recently playing with langium and was manage do make running LSP you can see it in branch `features/lsp`. This works fine for may little syntax file in `src/language-server/st.langium`. All that is needed now if only make that file right.

Unfortunately, it is a complicated process of testing it. It takes too long. There is a generator and if that passes ok, then it should work. I'll be working on that file, but your help will be appreciated. As I noticed some of you are familiar with EBNF file format.