Serhioromano/vscode-st #45: Add support for lowercase keywords

AndersHogqvist, 2022-09-21:
Even though we are supposed to use capital letters in keywords such as IF, ELSIF etc we have a lot of files where these are written in lower case (if, elseif...). However the syntax highlighting only check for these words in capital letters.

AndersHogqvist, 2022-09-21:
The only thing that needs to be done is to add `(?i)` at the beginning of all regex patterns in st.tmLanguage.json. Example:
`"match": "(?i)\\b(?:END_)?(?:IF|CASE|OF|ELSE|ELSIF|THEN|__TRY|__CATCH|__FINALLY|__ENDTRY)\\b"`

Serhioromano, 2022-09-23:
Or you can click Shift+Alt+F and this extension will capitalize all needed keywords. This way is much better because although ST is not case sensitive, it would be nicer to have those keywords capitalized for code convention.

AndersHogqvist, 2022-09-23:
> Or you can click Shift+Alt+F and this extension will capitalize all needed keywords. This way is much better because although ST is not case sensitive, it would be nicer to have those keywords capitalized for code convention.

Oh, didn't know about that. Thanks!

closed 2022-09-23 by AndersHogqvist

Serhioromano, 2022-09-23:
@AndersHogqvist  Please I will very appreciate if you rate this extension here
https://marketplace.visualstudio.com/items?itemName=Serhioromano.vscode-st&ssr=false#review-details