Serhioromano/vscode-st #47: Variable with name containing TO is interpreted as a function

AndersHogqvist, 2022-10-06:
When a variable name contains `TO` the syntax highlighting will interpret and highlight it as a function.

Serhioromano, 2022-10-17:
Can you show me an example I cannot reproduce. Also refer to theme you are using.

AndersHogqvist, 2022-10-18:
I'm using the dark theme (of course ;-)) and here's some variables we have in one of the files:
https://app.box.com/s/i9thn748hh75fwj0xakesicw62ze2nnh

As you can see the first three variables are highlighted as functions and the rest as variables and that's because they contain _TO_

Serhioromano, 2022-10-20:
Now I see what you mean. it is not `TO` but `_TO_` the reason. I'll try to fix it.

Serhioromano, 2022-10-20:
I have released 1.12.14 version where I fixed number of things discovered because of your issue. Should be fine now.

closed 2022-10-20 by Serhioromano

Serhioromano, 2022-10-23:
@AndersHogqvist please report if it works fine now.

AndersHogqvist, 2022-10-23:
Yes sir, it works perfectly fine 🙂