Serhioromano/vscode-st #49: Nested comment and Syntax highlighting /coloring.

Ekristoffe, 2022-10-15:
Hello,
I have found if there is nested comment like:
```
(*
		bAddress				: BYTE;
		bLength				: BYTE; (* size of the data *)
		dwID				: DWORD; (* 32 bit long ID *)
*)
```
because of the first *), the rest of the commented code won't be seen as commented.
Could it be possible to fix this ?
Thanks

Serhioromano, 2022-10-17:
Perhaps there is a method but in Codesys it works the same. And in VS code in any other language comments like `/*` also works the same. So, I think it is how it supposed to work.

closed 2022-10-17 by Serhioromano

Ekristoffe, 2022-10-17:
I am sorry but in codesys 2.3 or 3.5 it work as intended.
Meaning each (* is considered as a level and the first *) doesn’t cancel all other comment level.
If you want I can make more screen capture of codesys 2.3 and 3.5 vanilla settings …
And yes this is a problem for codesys vanilla users.
I haven’t tried to use multi level /* */ in vs code yet…
I will try when back at home.

Serhioromano, 2022-10-18:
Looks like I have fixed it in 1.12.13 version. Please try it. And if you like it, please rate the extension.

Ekristoffe, 2022-10-19:
Thanks a lot, it work fine. and I already reviewed it (But using different account).