Serhioromano/vscode-st #52: Wrong line comment character

AndersHogqvist, 2022-12-02:
According to the IEC standard comments shall be surrounded by '(*' and '*)', this extension however uses '//' for line comments. If there are companies that deviate from the standard and actually use '//', perhaps this could be added as a configuration item?

PTKu, 2022-12-02:
FYI
All following is valid according to latest:

### Single line comments:
~~~
m:= 42; // comment for one line 
// a single line comment can start at
// the first character position. 
~~~
### Multiline comments
~~~
(* comment *)
(***************************
 Nice frame comment
****************************)
~~~

### Multi-line comment /* … */
~~~
/* comment in one 
or more lines */
~~~
### Nested comment (* .. (* .. *) ..*)
~~~
(* (* NESTING *) *)
~~~
### Nested comment /* .. /* .. */ .. */
~~~
/* /* NESTING */ */
~~~

in other words:

```
MULTI_LINE_COMMENT 
	:   ('(*' | '/*') .*? ('*)' | '*/')
  	;

SINGLE_LINE_COMMENT
    :   '//' ~[\r\n]*
    ;
```

AndersHogqvist, 2022-12-02:
Ok, didn't know that. We're working in OpenPCS from Kollmorgen/infoteam Software that only supports what seems like the first edition of the standard.

Since I can use comment blocks for lines as well I guess it's not worth it to add this as a configurable option. I'll close the ticket.

closed 2022-12-02 by AndersHogqvist