Serhioromano/vscode-st #56: END_METHOD, END_PROPERTY, SET, GET ??

TheColonel2688, 2023-08-30:
It doesn't seem like these Keywords are supported.

- END_METHOD
- END_PROPERTY
- SET (*For Property*)
- GET (*For Property*)

Serhioromano, 2023-09-04:
Please write an example of code with all those keywords. How that looks?

Serhioromano, 2023-11-19:
@TheColonel2688 I would add it if you would add examples of syntax for GET and SET. Is there END_GET? 

```
SET : PropertyName

END_SET
```

Somthing like that?

Serhioromano, 2023-11-19:
Done.

closed 2023-11-19 by Serhioromano

TheColonel2688, 2024-02-13:
I am sorry for never commenting back.

I think the best syntax for this would be.

PROPERTY *Name* : *Type*
    SET
    END_SET
    GET
    END_GET
END_PROPERTY

I think this would be the closest to Codesys syntax. Codesys doesn't have a text-only format defined for Properties as of right now, But I assume they will have one for Codesys GO!

Serhioromano, 2024-02-14:
Ok, I have added it according to your example. v1.12.25. Let's see if something change I'll update.