Serhioromano/vscode-st #57: Exclude files/directories from st language support

rbastens, 2023-09-06:
Is it a possibility to add a setting to exclude the ST language support from different files or directories.

Serhioromano, 2023-09-11:
This extension does not parse files that are not opened. Please explain your case. You have file extension that match this extension?

Serhioromano, 2023-11-19:
You have an option right now in the status bar to change file language and switch what language support extension will handle it. I'll look into it that is possible to add exclude folder.

Serhioromano, 2023-11-19:
You have an option right now in the status bar change file language and switch what language support extension will handle it. Looks like there is no way. No easy way. It is not clear to me how to do that. Because VS Code do it all itself. I only set a file extension. SO fo you only way is to switch language.

closed 2023-11-19 by Serhioromano