Serhioromano/vscode-st #61: Add option in extension configuration to change or add supported extensions

prutonis, 2024-06-12:
I work on some programs for [AX9](https://logicio.com/rtcu_obsolete.htm#AX9T) PLC and the extension of program files is **.vpl** . It will be nice that the extension could be configured to support highlight for different file extensions.

Serhioromano, 2024-07-26:
I can add this extension. Please attach example file I want to see a content.
Also note you can set any file for ST highlight. In status bar select language and choose Structured text.

Serhioromano, 2024-09-29:
> I work on some programs for [AX9](https://logicio.com/rtcu_obsolete.htm#AX9T) PLC and the extension of program files is **.vpl** . It will be nice that the extension could be configured to support highlight for different file extensions.

Is this a simple text file? I cannot add support if I do not have example file.

prutonis, 2024-10-17:
Here is an example of a VPL program:

```
INCLUDE rtcu.inc
// Uncomment math.inc to add math library support.
INCLUDE math.inc
INCLUDE pressens.inc
INCLUDE utils.inc

#DEFINE PHONE_ADMIN_1 "01234567"
#DEFINE PHONE_ADMIN_2 "24567890"

// Structs
STRUCT_BLOCK SbMqtt
   h            : INT;
   rc           : INT;
   ccnt         : SINT:=0;
   dcnt         : SINT:=0;
   scnt         : SINT:=0;
   status       : BOOL;
   isFallback   : BOOL := FALSE;
END_STRUCT_BLOCK;

STRUCT_BLOCK SbLog
   fd          : FILE;
   lineCt      : INT;
   readPos     : DINT := 0;
END_STRUCT_BLOCK;

//  Input variables that can be configured via the configuration dialog (These are global)
VAR_INPUT
    Ain1           : INT;       | Analog input pressure readings 4-20mA 
    Switch1        : BOOL; 
    Switch2        : BOOL;
    Switch3        : BOOL;
END_VAR;

//  Output variables that can be configured via the configuration dialog (These are global)
VAR_OUTPUT
    LED            : BOOL;      | LED that shows the program scan
    Aout4          : INT;        
END_VAR;

//  These are the global variables of the program
VAR
   psns           : PressureSensor;
   logsb          : SbLog;
   clock          : clockGet;
   CFG            : ARRAY[1..16] OF STRING;
   call           : gsmIncomingCall;           // Receives incoming Voice call
   sms            : gsmIncomingSMS;            // Receives incoming SMS messages
   stack          : array [1..50] of string;
   stacki         : int := 1;
   PJS            : ParseJSON;
   keys           : ARRAY[1..K_SIZE] OF STRING := KN_CID, KN_MQ_HOST, KN_MQ_HOST_FB, KN_MQ_USER, KN_MQ_PORT, KN_SEND_INTERVAL, KN_INACT_INTERVAL;
   mqbuf          : ARRAY[1..1024] OF SINT;
   mq             : SbMqtt;
   rxd            : mqttReceive;
   netInfo        : netGetInformation;
   iface          : SINT := 1; // 1 - Mobile data, 2 - LAN
   tim3           : TON;
   inactivityTime : DINT;
END_VAR;

FUNCTION ProcessConfig;
VAR
   status         : BOOL;
   rc             : INT;
END_VAR;
   status := LoadConfig();
[... 354 lines cut ...]
         Flog(msg:="Inactivity retries exceeded limit. Resetting the board.");
         boardReset(); // Resets the RTCU Unit
      ELSE
         MqttDisconnect();
         MqttConnect();
         Flog(msg:="Inactivity time reached. Resetting mqtt connection.");
      END_IF;
      tim3(trig:=ON);
   END_IF;
   rc := mqttWaitEvent(timeout := 1);
   IF rc > 0 THEN
      rxd();
      mq.scnt := 0;
      Flog(msg:="MQTT received data: "+strFromMemory(src := ADDR(mqbuf), len := rxd.size));
      ProcessIncomingJson(str:=strFromMemory(src := ADDR(mqbuf), len := rxd.size), topic := rxd.topic);
   ELSIF rc < 0 THEN //no connection, try to resubscribe
      Flog(msg:="MQTT RX error: "+intToStr(v:=rc));
      MqttResubscribe();
   END_IF;

   boardWatchdog(); // Reset the watchdog timer
END;

END_PROGRAM;
```

Serhioromano, 2024-10-21:
I am adding it. 

`INCLUDE` and `#DEFINE` is not quite ST syntax. Is it part of ST for AX9?

Serhioromano, 2024-10-22:
done

closed 2024-10-22 by Serhioromano

prutonis, 2024-10-22:
> I am adding it.
> 
> `INCLUDE` and `#DEFINE` is not quite ST syntax. Is it part of ST for AX9?

#DEFINE and INCLUDE are preprocessing commands like in C/C++.
More information could be found on their website:
https://www.logicio.com/HTML/index.html