Serhioromano/vscode-st #64: Comments of type '(*' do not take effect  if followed by a "("

raokun, 2024-12-10:
Comments of type '(*' do not take effect  if followed by a "(",like this.
[image]

Serhioromano, 2024-12-27:
There should be a whitespace. The simplest way is to press Shift+Alt+F for formatting and it will insert whitespace there.

raokun, 2025-01-22:
> There should be a whitespace. The simplest way is to press Shift+Alt+F for formatting, and it will insert whitespace there.
> 
> Anyway, thank you. I'll have a look into it to see what I can do.

Sry，wo try to to press Shift+Alt+F for formatting, but it not work,like this.

[image]

katarzynagorska, 2026-03-05:
Hi, I think i found similar issue with strings containing "(" - not sure if related or something different. You can see example in the picture below:

[image]

xjh22222228, 2026-03-11:
There is a similar issue here, if the comment is  `//测试 (V) `, the comment does not take effect because it contains non English and is enclosed in parentheses

Serhioromano, 2026-09-03:
Fixed

closed 2026-09-03 by Serhioromano